Plasmatree/PID-Analyzer #1: CMD close

papayou66, 2017-09-18:
Hello the CMD close when i try to read a log 
Attached the log 
[1.zip](https://github.com/Plasmatree/PID-Analyzer/files/1311116/1.zip)

Plasmatree, 2017-09-18:
Which system are you on? 

I got no problems with your log:
[image]

papayou66, 2017-09-18:
windows 10 64 b

apeelme, 2017-09-18:
Me too...i use _example_smooth_tune.BBL_, w10 home 64bit.
[image]

Plasmatree, 2017-09-18:
Ah, quite a hint! Thanks :)

Plasmatree, 2017-09-19:
Seems like Windows sometimes returns the input path with or without quote marks. Maybe different language settings cause this.  

I added a check for quotemarks. 
New version is up!

papayou66, 2017-09-19:
I found the issue !!! my bad:
You need to put the log in the same folder 
Know i need to learn how to use the diagram 
[image]

Plasmatree, 2017-09-19:
I'm not sure why your response for low input is below 1. Could be that it's caused by your deadband setting. I will have a look at it. 
What matters most is the shape. The actual hight is more comparable to stick sensitivity/rate (but not the same).

Plasmatree, 2017-09-19:
@apeelme Have you tried again?

apeelme, 2017-09-19:
@Plasmatree Thanks, working! ;)

closed 2017-09-19 by Plasmatree