Plasmatree/PID-Analyzer #2: cmd crashes

muhkuh0815, 2017-09-18:
when i try with the "example_smooth_tune.BBL"  example file (or any of my own BBL files)
the pid-analyzer crashes while trying to generate the first .png file

Reading log 0...   Done!
Processing:
roll...   Done!
pitch...  Done!
yaw...  Done!

following files get created in the directory

example_smooth_tune_temp1.BBL gets created (2mb)
example_smooth_tune_temp1.01.csv gets created (15mb)
example_smooth_tune_temp1.01.event gets created (154 bytes)
example_smooth_tune_0.png gets created  ....  with a filesize of 0 

do i just need some sort of Plot/gnuplot for windows ? or do i have to ditch my windows 7 PC for a new one ...

ps: thanks for the nice work you done with this tool.

Plasmatree, 2017-09-18:
It should be completely standalone. I guess I need to try it on win7. 
Thanks for the feedback!

muhkuh0815, 2017-09-19:
thank you for looking into it.
[image]

Plasmatree, 2017-09-19:
Reproduced it on Win7 and should be fixed now. 
New version is online!

Thanks for your feedback!

muhkuh0815, 2017-09-19:
works like a charm now.
thanks for the great work!

closed 2017-09-19 by muhkuh0815