Plasmatree/PID-Analyzer #3: [wiki or help manual]

papayou66, 2017-09-20:
Hey can we work on wiki or a little manual ? 
Thanks

Plasmatree, 2017-09-20:
It's planned. I just need some time to do it.

papayou66, 2017-09-20:
I can help ....

Plasmatree, 2017-09-20:
Ah, great! Out of curiosity, what's your background? 
I guess I will just start the wiki and structure it first. For further content any input is appreciated!

How does this work then? Are there pull-requests for the wiki?

papayou66, 2017-09-20:
best way by mail ? or slack

cachenbach, 2017-10-14:
This is an amazing tool! Needed only a few seconds to realize that my P-Term ist way to low.
But a manual would be great.
Thanks!!

papayou66, 2017-10-15:
[image]
What about this log ?

cachenbach, 2017-10-15:
Looks better than mine:
[image]
I dont know why you have a blue and a orange line in the section "strength", but the blue line could indicate a low P-Term?!

cachenbach, 2017-10-18:
I think you have two lines because of:
> If D Setpoint Transition is set in Betaflight, your tune and thus the response will differ for high RC-inputs. This fact is respected by calculating separate responses for inputs above and below 500 deg/s. With just moderate input, you will get one result, if you also do flips there will be two.

BentyFPV, 2018-03-01:
I'm having trouble working out how to interpret the results and adjust my PIDs. The PIDs overlaid on the bottom graphs are what i have already set - how do I know what needs to change?
[image]

Plasmatree, 2018-03-02:
Hi Benty,
it looks like you worked the throttle more than anythig else. That's not best practice for this tool: It analyses the connection of intended movement and resulting movement on roll, pitch and yaw. Try to work these axis more than throttle. I usually do some moves left/right (for roll), front/back (pitch) and some yaw. Flips will produce a second graph representing the response for high input levels. 
There are no crazy moves required, it actually also works in angle mode! :)

Now for your results: Stock tune should be pretty good already for 4"/5".  Try increasig p and look at the effect (the edge will get sharper and a bump will appear). Increasing D suppresses the bump.  As long as your motors don't get too hot or oscillations emerge, this will yield a faster rise/quicker response.

On another note: Your quad seems to have some couplig of throttle and especially roll. Is it heavier on one side? Are motors/escs/props ok?

BentyFPV, 2018-03-02:
Thanks heaps for the reply.

Truth be told I discovered this tool thanks to Joshua Bardwell mentioning
it on a recent livestream. I jumped on the pc and grabbed it straight away
without collecting any new logs since I was super keen to give it a try.
Thats just the first log file I found, so its not a good example to use I
know. Its a friends quad Im tuning for him and I suspect there might be a
gyro noise issue as its a bit jittery and the motors get hot with
relatively low D gains... thats a different issue though :)

Ive been debating with friends all day the meaning of the graphs... am I
right in saying when tuning we're aiming to increase P for the steepest
possible initial slope up and increase D to the flattest possible peak?
(While keeping an eye on motor temps of course).

Am I right in guessing the graphs represent a sort of averaged or typical
pid sum value for each axis after your typical setpoint change?

Is there anything to be gleaned about the I term here? Ir is that really
just about personal preference?

My

Plasmatree, 2018-03-02:
Mind sharing the link to the video? Haven't noticed!

It all dempends on what your goal is. The perfect, neutral tune would be an instant, vertical rise to 1 and then staying there. That's of course not possible in reality but getting as close without frying your motors should give you the most locked in/neutral tune. 
Also test the influence of I: It usually produces a kind of delayed bump when set too high.

The graphs represet some kind of average but the current method is not perfect. There will be an update soon that does a better job at selecting the right data.

Zuldan, 2018-03-02:
@Plasmatree looking forward to the update! Can't wait ;-)

BentyFPV, 2018-03-02:
The JB video is here, and the mention of PID Analyzer is at about 43
minutes.

https://youtu.be/j_VZ0mBYXK0

Plasmatree, 2018-03-22:
There is some wiki now and I will be adding more. 
Guess it can be closed ;)

closed 2018-03-22 by Plasmatree