Plasmatree/PID-Analyzer #4: Tight Layout error?

apeelme, 2017-09-20:
When click _Tight layout_ button:
[image]
[image]

my log: 
[LOG00001.zip](https://github.com/Plasmatree/PID-Analyzer/files/1318745/LOG00001.zip)

Plasmatree, 2017-09-26:
I guess it's caused by GridSpec which is used to arrange the plots. 
Is it crutial for you? It has nothing to do with the data, just with the plot arrangement within the window.

apeelme, 2017-09-26:
I do not use, just tried it, and I wanted to help. ;)

Plasmatree, 2017-09-27:
Thanks, I will keep it in mind!

closed 2017-09-27 by Plasmatree