Plasmatree/PID-Analyzer #5: provide cli only interface

airtonix, 2018-03-05:
Something that doesn't require X11 or WinGDIBlahFoo.

Then we can put it in docker and now from end users point of view, docker is the runtime not python. cross platform problem avoided.

Zuldan, 2018-03-06:
Another option would be to convert everything to Powershell Core (6.0) and then it would run natively on any platform (I.E. not running Python virtual environment).

Plasmatree, 2018-04-13:
Version 0.3 has climate mode now.

closed 2018-04-13 by Plasmatree

airtonix, 2018-04-24:
@Plasmatree windows users. depends on the install.

But From my years of experience with python and nodejs. I assure you that installing docker for windows or even just docker on linux is much easier than installing python (then pip installing the right things) or installing nodejs (then trying to get all the 3rd and 4th party deps right).

So the docker suggestion is not just about cross platform, it's mostly about shielding your users from the problems of installing it. The idea is that you do it for them, they just run the black box that is your docker image.

As to the concern about requiring virtualisation:

1. moby linux is tiny. the integration between host and docker in the vm is seemless (i use it every day in my day job to remove the complexity of installing tooling)
2. you can do what is called a multi arch docker file so that windows-docker (not docker on mobyvm on windows) will use the windows arch base image, and docker on linux will pull the linux arch, and docker on arm will pull the arm base image.
  - https://blog.docker.com/2017/11/multi-arch-all-the-things/
  - https://lobradov.github.io/Building-docker-multiarch-images/