Plasmatree/PID-Analyzer #7: suggest add INAV log support

sincoder, 2018-04-16:
failed to analyze INAV BB logs , hope it could be supported
https://github.com/iNavFlight/blackbox-log-viewer

sincoder, 2018-04-16:
[LOG00003.zip](https://github.com/Plasmatree/PID-Analyzer/files/1914344/LOG00003.zip)
INAV log example

sincoder, 2018-04-16:
problem cause by there is no axisD[2] in INAV logs , just set it to 0

Plasmatree, 2018-04-16:
That's correct. Skipping on axisD[2] (that's yaw) now, since there is never D on yaw.

Thank's for the log. I suggest stronger movements, to get more relaiable data. 

Would you mind giving some detail on your setup? 
Seem's to be a class I don't have experience with.

sincoder, 2018-04-16:
what are you mean setup ? my quad or INAV fc ? quad was matek F405 CTR + M8N GPS

Plasmatree, 2018-04-16:
Your quad. Is it a q380? 
Props/Motors/weight/battery?

Just to get a feel for the numbers and how it relates to the system response.

sincoder, 2018-04-16:
yes , it is Q380 , without battery it about 590g  ,phantom 3 motor, 9450 prop ,frame q380, 20A ESC  , battery 4S 4200mhA  433g

closed 2018-04-16 by Plasmatree

rosemike23, 2024-05-15:
Hi, will removing the  axisD[2] will work for INAV TXT ?