Plasmatree/PID-Analyzer #8: windows versions not run

fmartinezo, 2018-04-17:
I can not run any of the available windows versions on windows 10. The same error always comes out. A small blue screen that says "this application can not be run on the computer, check with the software editor"

Plasmatree, 2018-04-18:
Are you running 32 bit?
The current versions are 64bit.

fmartinezo, 2018-04-18:
Yes i am running 32 bit

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Libre
de virus. www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

fmartinezo, 2018-04-18:
May be possible to obtain w32 version?

Plasmatree, 2018-04-18:
I already compiled one, but there's a bug. Will have a look at it.

fmartinezo, 2018-04-18:
Thank so much, pls tell me when it will be available

Aileron8, 2018-04-19:
cmd closes for me when dropping logs. I'm running a win7 64 bit pc. Decode is in the same folder and works. - I just want to add that this program looking amazing. So hoping to get it up and running for sure.

Plasmatree, 2018-04-19:
Does it close with wisth a single file input? 
What is left in the /tmp folder afterwards?

Aileron8, 2018-04-19:
Yes, it closes with a single file. The first issue is the PID-Analyzer isn't executing properly. The cmd closes right away there as well and the tmp folder aren't being generated. I'm using Python 2.7. Could that be the issue?

Plasmatree, 2018-04-19:
You actually don't need python to run the exe, everything is included. 
If you call it via cmd the window won't close and you can better read what's going on.

How does your path to the folder look? Maybe put the folder to a simple location like "c:"and try again.

Aileron8, 2018-04-20:
I mean, because I have Python installed, perhaps it's making things more complex? When I try to run the exe from cmd or within IDLE. 

`Traceback (most recent call last):
  File "C:\PID-Analyzer-master\PID-Analyzer.py", line 7, in <module>
    import numpy as np
  File "C:\PID-Analyzer-master\numpy\__init__.py", line 131, in <module>
ImportError: Error importing numpy: you should not try to import numpy from
        its source directory; please exit the numpy source tree, and relaunch
        your python interpreter from there.`

Aileron8, 2018-04-20:
When I "exit the numpy source tree, and relaunch your python interpreter from there." numpy is not found and an error is generated at line 7 of the program as a result.

Plasmatree, 2018-04-20:
That's stange. I'm also running 2.7 and there doesn't seem to be any interaction.

If you are familiar with python anyways, you can also run the .py to get it working at least.

Aileron8, 2018-04-20:
That was the first thing I tried when it wasn't functioning, but I think I found the issue.  If I'm unable to get it working on my pc I'm sure my mac will not have this issue since I'm the admin. Thank you for your assistance. I'm quite eager to analyze some charts and tune like never before. 

https://stackoverflow.com/questions/14570011/explain-why-numpy-should-not-be-imported-from-source-directory

closed 2018-05-21 by Plasmatree