Plasmatree/PID-Analyzer #9: Are the new BF features going to function?

Aileron8, 2018-04-22:
I was able to get things working in my mac. The tool is great! I have a question though? Recently there have been a few additions in BF that impact the PIDs. Is it fair to say the PID-Analyzer simulates the PID controller when tabulating the step function charts from the bb log data? Would the recent Iterm_rotation, Throttle_boost, and other changes be accounted for in the tool?

Plasmatree, 2018-04-22:
The only thing it's depending on is the relation of P-Value, P-error and gyro. 
As long as the p-term doesn't change it shouldn't affect the program, since there is no simulation at the moment (I got some ideas for simulation based auto tuning, but that's a different topic... ;) ). 

Keep in mind that it breaks down the complex behaviour of the drone to a single curve. This only works by assuming relatively linear behaviour. Accounting for further correlations to dynamic features is something for the future (except for the additional results for high input rates which is implemented exactly for this reason). 
Theoretically the information is there, but extracting and presenting it as an easily understandable result is the hard part.

If you think you see an influence of the dynamic features on the results, please share the logs! 
It will help to further improve on tool.

Vidalcris, 2018-06-11:
I dont see the point about those new feature related to PID-analyzer :-/
I mean this tool will just work exactly the same ^^ Throttle boost change the throttle but this is exactly the same when you move the throttle without throttle boost ...

I'm already using this tool with all the latest features  : smart_feedforward, iterm_rotation, throttle_boost, iterm_relax, rc_smoothing !

Here is the result : 
[image]
[image]

So yeah those features allow me to have this pretty nice response curve, and yes the tool show the effect of the features already :)

Now correct me if there is something i misunderstood.

Aileron8, 2018-07-06:
I didn’t see your response until now. At the time I wrote this post back in April throttle boost and/or high throttle, high noise logs we’re giving me different results. This no longer the case. I now fly with all my features and settings on with great results. 8-)

Plasmatree, 2018-07-06:
OK, so it's closed :)

closed 2018-07-06 by Plasmatree