Plasmatree/PID-Analyzer #10: Compatibility with KISS

vever001, 2018-05-01:
Hello @Plasmatree ,

I just discovered this tool and it looks very interesting so thank you very much for this!
I'm mainly using a KISS FC V2 ATM but I ran into a few issues.
Here is my log (100% log rate, no debug set): 
[GOPR0173.TXT](https://github.com/Plasmatree/PID-Analyzer/files/1963812/GOPR0173.TXT)

I get the following error message:
`INFO 2018-04-30 20:33:14,848 PID-Analyzer.py:562: PID-Analyzer 0.31
INFO 2018-04-30 20:33:14,848 PID-Analyzer.py:563: Hello Pilot!
INFO 2018-04-30 20:33:14,848 PID-Analyzer.py:572: Interactive mode: Enter log file, or type close when done.
Balckbox log file path (type or drop here): LOG00015.TXT
Optional plot name:kisstest
WARNING 2018-04-30 20:33:23,625 PID-Analyzer.py:518: Ignoring BBL session '...\\test2\\LOG00015_temp0.TXT', 61B < 500000B.
Decoding log '...\LOG00015_temp1.TXT' to '...\test2\LOG00015_temp1.01.csv'...

Log 1 of 1, start 00:51.011, end 04:07.178, duration 03:16.166

Statistics
Looptime            999 avg           75.4 std dev (7.5%)
I frames    6085   36.9 bytes avg   224832 bytes total
P frames  187419   20.6 bytes avg  3854044 bytes total
E frames       2    9.0 bytes avg       18 bytes total
Frames    193504   21.1 bytes avg  4078876 bytes total
Data rate  986Hz  20943 bytes/s     209500 baud

92 frames failed to decode, rendering 1354 loop iterations unreadable. 2686 iterations are missing in total (2685ms, 1.37%)

INFO 2018-04-30 20:33:25,760 PID-Analyzer.py:308: Reading log 0...
INFO 2018-04-30 20:33:26,469 PID-Analyzer.py:328:       Done!
INFO 2018-04-30 20:33:26,470 PID-Analyzer.py:200: Processing:
Traceback (most recent call last):
  File "PID-Analyzer.py", line 591, in <module>
  File "PID-Analyzer.py", line 524, in run_analysis
  File "PID-Analyzer.py", line 364, in __init__
  File "PID-Analyzer.py", line 381, in _csv_iter
  File "PID-Analyzer.py", line 201, in __init__
  File "PID-Analyzer.py", line 339, in find_traces
ValueError: could not convert string to float:
Failed to execute script PID-Analyzer`

Plasmatree, 2018-05-01:
Oh, never tried kiss. Didn't know that the logs are compatible! 
Can you upload/send the log?

vever001, 2018-05-01:
@Plasmatree It's there in the description.
Thanks a lot ;)

Plasmatree, 2018-05-09:
I had a look at it, but it's not that straight forward. 
Problem is that I don't know the exact PID and RC-input  implementation of kiss since it's closed source. 

Differences to betaflight I noticed:
- The gyro scale is different
- the PID-scaling factors are unknown (~10x compared to betaflight?)  
- the PID-values are not saved in the log header

These are all necessary. 
I will search some more, or maybe even ask their support. 
Nonetheless thanks for the inspiration! ;)

vever001, 2018-05-09:
Hi @Plasmatree 
The PID scaling may very well be 10x for roll and pitch as the values I put there are quite similar to many people and it does fly very very well.
For yaw they probably have another scaling. There is an option called "yaw filter strength" in the GUI which is set by default to 35 (which is what I'm using). Maybe this impacts the scaling.
But yes, better ask them what's going on exactly ;)

ScorpionTDL, 2018-05-24:
Version 0.51 seems to work somehow with a Kiss V2 log file without the need to add anything manually to the lofgile.
The PID and TPA infos are missing as Kiss does not write those to the logfile. But the general Charts seem to look ok (as far as I can tell without having any real knowledge about this ;-) )
I tried some tuning and it seems to be working with my little understanding. at last Changes in P and D have the desired effects. Increasing I however does not move the lines up, instead it goes down. And I seems to have some dampening effects that I did not expect. Don't know if this is a Kiss specific thing though.

[image]
[image]
[LOG00074.TXT](https://github.com/Plasmatree/PID-Analyzer/files/2035200/LOG00074.TXT)

Plasmatree, 2018-05-24:
Haha, sweet! That wasn't really planned. :D 
The gyro scale is still different. 
What's you max rate? Something 1100deg/s?

ScorpionTDL, 2018-05-25:
Yea I'm running 1027° on roll and pitch and 70° on yaw.
I'm running the adaptive filter from kiss, so notch should normaly not be requiered. however I might give it a try just for fun ;-)

[image]

With this current tune I'm not quiet sure if D is not too high on pitch, as it seems as if D is oscilation quiet a bit when looking at the data with blackbix explorer:
[image]

Plasmatree, 2018-05-25:
If the dynamic notch is similar to bf, it fails >400hz. That's where your motor noise is amplified by the d-term (top right corner: 70-100% throttle, ~410hz). Put a notch on there, preferably in gyro and mybe skip on the d-term notch instead (it's even possible that the d-term notch supports a resonance at higher frequencies, since it induces additional phaseshift between P and D, but that's someting I'm currently investigating myself.)
It seems like you got a frame (?) resonance at 100hz on pitch (constant horizontal line). Maybe D on pitch is really slightly high, since also the reponse also looks a bit over damped.

toper74, 2018-07-31:
Hi ! so is this version kiss compatible without any add on the txt log ?