Plasmatree/PID-Analyzer #12: Program crash on windows

atomiclama, 2018-05-09:
Would like to give this program a spin :-)
But having an issue with the program crashing. Hope the attached info can help you track it down.

Running Windows 7 Enterprise SP1 64bit.

Anything else just shout and I'll see if I can help out.

[image]

c:\temp\blackbox>PID-Analyzer_0.31.exe
INFO 2018-05-09 09:18:38,266 PID-Analyzer.py:560: Decoding with 'c:\\temp\\black
box\\Blackbox_decode.exe'
INFO 2018-05-09 09:18:38,266 PID-Analyzer.py:562: PID-Analyzer 0.31
INFO 2018-05-09 09:18:38,266 PID-Analyzer.py:563: Hello Pilot!
INFO 2018-05-09 09:18:38,266 PID-Analyzer.py:572: Interactive mode: Enter log fi
le, or type close when done.
Balckbox log file path (type or drop here): test
Optional plot name:
INFO 2018-05-09 09:19:27,496 PID-Analyzer.py:593: No valid input path!
INFO 2018-05-09 09:19:27,496 PID-Analyzer.py:572: Interactive mode: Enter log fi
le, or type close when done.
Balckbox log file path (type or drop here): C:\temp\blackbox\BTFL_BLACKBOX_LOG_2
0180425_072557.BBL
Optional plot name:test
WARNING 2018-05-09 09:19:34,811 PID-Analyzer.py:518: Ignoring BBL session 'C:\\t
emp\\blackbox\\test\\BTFL_BLACKBOX_LOG_20180425_072557_temp0.BBL', 61B < 500000B
.
Decoding log 'C:\temp\blackbox\test\BTFL_BLACKBOX_LOG_20180425_072557_temp1.BBL'
 to 'C:\temp\blackbox\test\BTFL_BLACKBOX_LOG_20180425_072557_temp1.01.csv'...

Log 1 of 1, start 00:53.418, end 01:44.549, duration 00:51.130

Statistics
Looptime            511 avg           64.6 std dev (12.6%)
I frames    1537   41.6 bytes avg    63935 bytes total
P frames   96797   24.4 bytes avg  2361992 bytes total
E frames       2    9.0 bytes avg       18 bytes total
S frames      31    4.0 bytes avg      124 bytes total
Frames     98334   24.7 bytes avg  2425927 bytes total
Data rate 1923Hz  47507 bytes/s     475100 baud

1 frames failed to decode, rendering 15 loop iterations unreadable. 294927 itera
tions are missing in total (38345ms, 75.00%)

INFO 2018-05-09 09:19:36,582 PID-Analyzer.py:308: Reading log 0...
PID-Analyzer.py:364: DtypeWarning: Columns (31) have mixed types. Specify dtype
option on import or set low_memory=False.
INFO 2018-05-09 09:19:40,822 PID-Analyzer.py:328:       Done!
INFO 2018-05-09 09:19:40,822 PID-Analyzer.py:200: Processing:
INFO 2018-05-09 09:19:40,822 PID-Analyzer.py:302: roll...
INFO 2018-05-09 09:19:41,681 PID-Analyzer.py:302: pitch...
INFO 2018-05-09 09:19:42,411 PID-Analyzer.py:302: yaw...

[BTFL_BLACKBOX_LOG_20180425_072557.zip](https://github.com/Plasmatree/PID-Analyzer/files/1986950/BTFL_BLACKBOX_LOG_20180425_072557.zip)

Plasmatree, 2018-05-09:
Looks like a problem during plotting or saving. 
Thanks for the heads up!
This is what I get from your log:
[image]

atomiclama, 2018-05-09:
Yeah a .png file is created but is 0 size.

chmod755a, 2018-05-12:
I have the same problem. 

I'm running Windows 7 Profession SP1 64-bit.

I tried running as administrator and Windows 7 compatibility and it doesn't help.

[image]

Log attached.

[LOG00010.zip](https://github.com/Plasmatree/PID-Analyzer/files/1997267/LOG00010.zip)

jfinky, 2018-05-12:
I'm having the same issue with windows 7 Ultimate 64bit. 

`INFO 2018-05-12 20:59:24,421 PID-Analyzer.py:560: Decoding with 'C:\\Users\\blah
\\Downloads\\quad\\plasma tree\\Blackbox_decode.exe'
INFO 2018-05-12 20:59:24,423 PID-Analyzer.py:562: PID-Analyzer 0.31
INFO 2018-05-12 20:59:24,424 PID-Analyzer.py:563: Hello Pilot!
INFO 2018-05-12 20:59:24,424 PID-Analyzer.py:572: Interactive mode: Enter log fi
le, or type close when done.
Balckbox log file path (type or drop here): godmodeswamp.bbl
Optional plot name:test
WARNING 2018-05-12 20:59:39,032 PID-Analyzer.py:518: Ignoring BBL session 'C:\\U
sers\\blah\\Downloads\\quad\\plasma tree\\test\\godmodeswamp_temp0.bbl', 61B < 5
00000B.
WARNING 2018-05-12 20:59:39,033 PID-Analyzer.py:518: Ignoring BBL session 'C:\\U
sers\\blah\\Downloads\\quad\\plasma tree\\test\\godmodeswamp_temp1.bbl', 16395B
< 500000B.
Decoding log 'C:\Users\blah\Downloads\quad\plasma tree\test\godmodeswamp_temp2.b
bl' to 'C:\Users\blah\Downloads\quad\plasma tree\test\godmodeswamp_temp2.01.csv'
...

Log 1 of 1, start 00:52.145, end 04:49.477, duration 03:57.332

Statistics
Looptime             70 avg            1.2 std dev (1.7%)
I frames    6513   40.2 bytes avg   261790 bytes total
P frames  410316   24.5 bytes avg 10039138 bytes total
E frames       2    9.0 bytes avg       18 bytes total
S frames      27    4.0 bytes avg      108 bytes total
Frames    416829   24.7 bytes avg 10300928 bytes total
Data rate 1756Hz  43414 bytes/s     434200 baud

1 frames failed to decode, rendering 1 loop iterations unreadable. 1 iterations
are missing in total (0ms, 0.00%)
2917803 loop iterations weren't logged because of your blackbox_rate settings (2
07665ms, 87.50%)

INFO 2018-05-12 20:59:45,457 PID-Analyzer.py:308: Reading log 0...
INFO 2018-05-12 20:59:47,635 PID-Analyzer.py:328:       Done!
INFO 2018-05-12 20:59:47,641 PID-Analyzer.py:200: Processing:
INFO 2018-05-12 20:59:47,648 PID-Analyzer.py:302: roll...
INFO 2018-05-12 20:59:50,256 PID-Analyzer.py:302: pitch...
INFO 2018-05-12 20:59:53,203 PID-Analyzer.py:302: yaw...`

`Problem signature:
  Problem Event Name:	APPCRASH
  Application Name:	PID-Analyzer_0.31.exe
  Application Version:	0.0.0.0
  Application Timestamp:	00021856
  Fault Module Name:	ntdll.dll
  Fault Module Version:	6.1.7601.18869
  Fault Module Timestamp:	556366f2
  Exception Code:	c0000005
  Exception Offset:	000000000004ada4
  OS Version:	6.1.7601.2.1.0.256.1
  Locale ID:	3081
  Additional Information 1:	62fa
  Additional Information 2:	62fa78b31418813259dc77cb55c4a384
  Additional Information 3:	4168
  Additional Information 4:	41689d6a898badccec125e0065650d25

Read our privacy statement online:
  http://go.microsoft.com/fwlink/?linkid=104288&clcid=0x0409

If the online privacy statement is not available, please read our privacy statement offline:
  C:\Windows\system32\en-US\erofflps.txt
`
The crash occurs when it's generating the png file. 
I have attached two logs. The kaiser log runs perfectly fine yet the godmode one and several others cause the crash. I've tried running it as admin and under different compatibility modes but it makes no difference. 

[logs.zip](https://github.com/Plasmatree/PID-Analyzer/files/1997536/logs.zip)

atomiclama, 2018-05-17:
@Plasmatree ok tried the V0.4 and it now creates the .png files ok. But.....

It crashes when using the built in part of the program to display and navigate the .png files.

Anyway thanks for fixing so far.

Plasmatree, 2018-05-18:
The plots are a bit less resource hungy now, since I changed the content of the color plot in version 0.5.
How much ram do you have? Processor? Just as reference.

chmod755a, 2018-05-19:
Mine crashes at "Interactive mode: Enter log file, or type close when done".  If I answer "Y" to "show plot window when done" then it crashes right after the noise graph is plotted.

I have 24GB ram (3.5GB free) and my CPU is an Intel i7-2600K @ 3.40 GHz.

At least now the PNG files are created and I have something to look at.

atomiclama, 2018-05-21:
@Plasmatree memory and processor specs for reference.
[image]

Plasmatree, 2018-05-21:
@chmod755a: Is there a problem when you don't show the plots (press 'n')?

Lord-Quake, 2018-05-29:
Works fine on my machine (Windows 7 Professional 64bit)

Zuldan, 2018-05-29:
Weird. I did a fresh vanilla install on Win7 then ran PID Analyzer without installing anything and it crashes after generating the graphs.

Zuldan, 2018-05-30:
So here is the error message for me, finishes the graphs, prompts for the next one then crashes. Windows 7 Pro (SP1) x64, fresh install, nothing installed but PID Analyzer.

[image]

I've attached the BBL I used.

[HELIO.zip](https://github.com/Plasmatree/PID-Analyzer/files/2052369/HELIO.zip)

Lord-Quake, 2018-05-30:
I just tried your helio file and the program did not crash on my system. So it seems to be an isolated problem.

trunglp, 2018-05-30:
Today i use version 0.51
application is crashed

INFO 2018-05-30 14:29:40,823 PID-Analyzer_0.51.py:655: pitch...
INFO 2018-05-30 14:29:41,423 PID-Analyzer_0.51.py:655: yaw...
PID-Analyzer_0.51.py:114: RuntimeWarning: invalid value encountered in divide
PID-Analyzer_0.51.py:219: RuntimeWarning: invalid value encountered in divide
C:\Users\lptrung\AppData\Local\Temp\_MEI90~1\numpy\core\_methods.py:26: RuntimeWarning: invalid value encountered in reduce
PID-Analyzer_0.51.py:98: RuntimeWarning: invalid value encountered in less_equal
PID-Analyzer_0.51.py:99: RuntimeWarning: invalid value encountered in greater
C:\Users\lptrung\AppData\Local\Temp\_MEI90~1\numpy\core\_methods.py:29: RuntimeWarning: invalid value encountered in reduce
INFO 2018-05-30 14:29:41,937 PID-Analyzer_0.51.py:546: Making PID plot...
Traceback (most recent call last):
  File "PID-Analyzer_0.51.py", line 960, in <module>
  File "PID-Analyzer_0.51.py", line 916, in run_analysis
  File "PID-Analyzer_0.51.py", line 758, in __init__
  File "PID-Analyzer_0.51.py", line 775, in _csv_iter
  File "PID-Analyzer_0.51.py", line 354, in __init__
  File "PID-Analyzer_0.51.py", line 560, in plot_all_resp
  File "site-packages\matplotlib\pyplot.py", line 1598, in ylim
  File "site-packages\matplotlib\axes\_base.py", line 3429, in set_ylim
  File "site-packages\matplotlib\axes\_base.py", line 3040, in _validate_converted_limits
**ValueError: Axis limits cannot be NaN or Inf
Failed to execute script PID-Analyzer_0.51**

Plasmatree, 2018-05-30:
@Zuldan: I can't reproduce it with your file. An alternative would be to install python and run the source directly. If you install [anaconda with python 2.7](https://www.anaconda.com/download/), you got everything you need.
Then just use cmd, cd to your analyzer folder and run: "python pid-analyzer.py"

@trunglp: can you please post the log file?

trunglp, 2018-05-31:
This is my log
https://drive.google.com/file/d/1FWiqzPy4Id0Jz6PKx-2CSKmcNWOuguBU/view?usp=sharing

atomiclama, 2018-06-06:
Version 0.51 is working fine now. Thanks.
@Plasmatree Shall I keep this issue open for others to report or close?

atomiclama, 2018-06-08:
Ok I take that back V0.51 still crashes, produces plots but then crashes.

V0.52 locks up 

`INFO 2018-06-08 10:42:09,006 PID-Analyzer_0.52.py:982: Decoding with 'D:\\Andy\\
quad\\log\\tuning\\Blackbox_decode.exe'
INFO 2018-06-08 10:42:09,006 PID-Analyzer_0.52.py:984: PID-Analyzer 0.52
INFO 2018-06-08 10:42:09,006 PID-Analyzer_0.52.py:985: Hello Pilot!
INFO 2018-06-08 10:42:09,006 PID-Analyzer_0.52.py:999: Interactive mode: Enter l
og file, or type close when done.
Blackbox log file path (type or drop here): D:\Andy\quad\log\tuning\2008-06-08\L
OG00021.BFL
Optional plot name:
Show plot window when done? [Y]/Nn
Bounds on noise plot: [default/last] | copy and edit | "auto"
Current: [[1.0, 10.1], [1.0, 100.0], [1.0, 100.0], [0.0, 4.0]]
`

Plasmatree, 2018-06-09:
Hit enter to continue.

atomiclama, 2018-06-11:
Doh :-)
 not that obvious though.

Sorry still crashes 
[image]

atomiclama, 2018-06-11:
Ok Narrowed it down a bit. The crash is happening when run in interactive mode when I drag and drop the  file.

If I run and specify the log files on the command line then it is fine.

Hope that helps, shout if you want need anything else.

Plasmatree, 2018-06-12:
What's the exact name of the file in questions?

atomiclama, 2018-06-12:
There are no spaces in the file name or directory path. I think that was just a problem coming from the copy from windoze cmd terminal.

I have re-run the tests, this is specifying the log file from the command prompt: -
`D:\Andy\quad\log>PID-Analyzer_0.52.exe -s N -l 2018-06-09\LOG00029.BFL
INFO 2018-06-12 09:44:10,131 PID-Analyzer_0.52.py:982: Decoding with 'D:\\Andy\\
quad\\log\\Blackbox_decode.exe'
INFO 2018-06-12 09:44:10,131 PID-Analyzer_0.52.py:984: PID-Analyzer 0.52
INFO 2018-06-12 09:44:10,132 PID-Analyzer_0.52.py:985: Hello Pilot!
WARNING 2018-06-12 09:44:10,171 PID-Analyzer_0.52.py:933: Ignoring BBL session '
D:\\Andy\\quad\\log\\2018-06-09\\tmp\\LOG00029_temp0.BFL', 61B < 500000B.
Decoding log 'D:\Andy\quad\log\2018-06-09\tmp\LOG00029_temp1.BFL' to 'D:\Andy\qu
ad\log\2018-06-09\tmp\LOG00029_temp1.01.csv'...

Log 1 of 1, start 01:43.306, end 02:41.091, duration 00:57.785

Statistics
Looptime           1003 avg          193.4 std dev (19.3%)
I frames    1731   43.1 bytes avg    74581 bytes total
P frames   53595   24.6 bytes avg  1320924 bytes total
E frames       2    9.0 bytes avg       18 bytes total
S frames       8    4.0 bytes avg       32 bytes total
Frames     55326   25.2 bytes avg  1395505 bytes total
Data rate  957Hz  24239 bytes/s     242400 baud

6 frames failed to decode, rendering 82 loop iterations unreadable. 390912 itera
tions are missing in total (50620ms, 87.60%)

INFO 2018-06-12 09:44:11,158 PID-Analyzer_0.52.py:676: Reading: Log 0
INFO 2018-06-12 09:44:11,404 PID-Analyzer_0.52.py:356: Processing:
INFO 2018-06-12 09:44:11,405 PID-Analyzer_0.52.py:671: roll...
INFO 2018-06-12 09:44:13,029 PID-Analyzer_0.52.py:671: pitch...
INFO 2018-06-12 09:44:14,535 PID-Analyzer_0.52.py:671: yaw...
INFO 2018-06-12 09:44:16,045 PID-Analyzer_0.52.py:573: Making PID plot...
INFO 2018-06-12 09:44:17,530 PID-Analyzer_0.52.py:664: Saving as image...
INFO 2018-06-12 09:44:19,385 PID-Analyzer_0.52.py:377: Making noise plot...
INFO 2018-06-12 09:44:19,937 PID-Analyzer_0.52.py:564: Saving as image...
INFO 2018-06-12 09:44:21,986 PID-Analyzer_0.52.py:940: Analysis complete, showin
g plot. (Close plot to exit.)`

This is running the programme then dragging and dropping the same log file: -
`D:\Andy\quad\log>PID-Analyzer_0.52.exe
INFO 2018-06-12 09:47:47,000 PID-Analyzer_0.52.py:982: Decoding with 'D:\\Andy\\
quad\\log\\Blackbox_decode.exe'
INFO 2018-06-12 09:47:47,000 PID-Analyzer_0.52.py:984: PID-Analyzer 0.52
INFO 2018-06-12 09:47:47,002 PID-Analyzer_0.52.py:985: Hello Pilot!
INFO 2018-06-12 09:47:47,002 PID-Analyzer_0.52.py:999: Interactive mode: Enter l
og file, or type close when done.
Blackbox log file path (type or drop here): D:\Andy\quad\log\2018-06-09\LOG00029
.BFL
Optional plot name:
Show plot window when done? [Y]/Nn
Bounds on noise plot: [default/last] | copy and edit | "auto"
Current: [[1.0, 10.1], [1.0, 100.0], [1.0, 100.0], [0.0, 4.0]]

WARNING 2018-06-12 09:48:07,910 PID-Analyzer_0.52.py:933: Ignoring BBL session '
D:\\Andy\\quad\\log\\2018-06-09\\tmp\\LOG00029_temp0.BFL', 61B < 500000B.
Decoding log 'D:\Andy\quad\log\2018-06-09\tmp\LOG00029_temp1.BFL' to 'D:\Andy\qu
ad\log\2018-06-09\tmp\LOG00029_temp1.01.csv'...

Log 1 of 1, start 01:43.306, end 02:41.091, duration 00:57.785

Statistics
Looptime           1003 avg          193.4 std dev (19.3%)
I frames    1731   43.1 bytes avg    74581 bytes total
P frames   53595   24.6 bytes avg  1320924 bytes total
E frames       2    9.0 bytes avg       18 bytes total
S frames       8    4.0 bytes avg       32 bytes total
Frames     55326   25.2 bytes avg  1395505 bytes total
Data rate  957Hz  24239 bytes/s     242400 baud

6 frames failed to decode, rendering 82 loop iterations unreadable. 390912 itera
tions are missing in total (50620ms, 87.60%)

INFO 2018-06-12 09:48:08,948 PID-Analyzer_0.52.py:676: Reading: Log 0
INFO 2018-06-12 09:48:09,194 PID-Analyzer_0.52.py:356: Processing:
INFO 2018-06-12 09:48:09,194 PID-Analyzer_0.52.py:671: roll...
INFO 2018-06-12 09:48:10,802 PID-Analyzer_0.52.py:671: pitch...
INFO 2018-06-12 09:48:12,296 PID-Analyzer_0.52.py:671: yaw...
INFO 2018-06-12 09:48:13,766 PID-Analyzer_0.52.py:573: Making PID plot...
INFO 2018-06-12 09:48:15,359 PID-Analyzer_0.52.py:664: Saving as image...
INFO 2018-06-12 09:48:17,292 PID-Analyzer_0.52.py:377: Making noise plot...
INFO 2018-06-12 09:48:17,864 PID-Analyzer_0.52.py:564: Saving as image...
INFO 2018-06-12 09:48:19,887 PID-Analyzer_0.52.py:940: Analysis complete, showin
g plot. (Close plot to exit.)
INFO 2018-06-12 09:48:19,943 PID-Analyzer_0.52.py:999: Interactive mode: Enter l
og file, or type close when done.
Blackbox log file path (type or drop here):`

~1sec later the this appears: -
[image]