Plasmatree/PID-Analyzer #14: Just question

Legista, 2018-05-10:
I am sorry for writing this in Issues, but i have no idea how to ask a question about this amazing Analyzer.
I just can't understand how to interpret plot "xxx step response (>500)".
What is 500? Is it settling time in ms? What values are present in this plot and how they correlate with X axis and times on this axis. I just can't get the point.
Thank you!
[image]

Legista, 2018-05-10:
Oh, it seems like i got it!.
it's rate, right? I.e. blue line for rates lower than 500 and yellow for rates over 500 (deg/s?).

Plasmatree, 2018-05-10:
:D yep, you got it right!
You can probably increase D on roll to suppress the overshoot. 
And maybe up P on pitch for a sharper edge, but that's hard to tell.

Legista, 2018-05-11:
I am starting to get a handle on this, but only just the beginnings.
Plasmatree, your program is awesome, thank you very much.

Zuldan, 2018-05-11:
https://youtu.be/a0E1ua_ASCQ