Plasmatree/PID-Analyzer #15: How to Use - MacOSX

Bradw-17, 2018-05-10:
HOW TO RUN PLASMATREE (MAC-OSX)

#Install: 
Python 2.7(https://www.python.org/downloads/release/python-2715/
PIP: (https://pip.pypa.io/en/stable/installing/)
Homebrew + Git: (https://gist.github.com/derhuerst/1b15ff4652a867391f03…)
Xcode: (https://itunes.apple.com/us/app/xcode/id497799835)

#Open Terminal - Do:

git clone https://github.com/cleanflight/blackbox-tools.git

cd blackbox-tools

brew install cairo --without-x11 pkg-config

make

cd ..

git clone https://github.com/Plasmatree/PID-Analyzer.git

cd PID-Analyzer

pip install -r requirements.txt

#Normal Use in Terminal:

cd PID-Analyzer

python PID-Analyzer.py --blackbox_decode ../blackbox-tools/obj/blackbox_decode -l (path_to_the_file.BBL)

#path_to_the_file.BBL > Drag and drop log to TERMINAL window

#Hit enter and boom!

Credit: Martin Hapl

Plasmatree, 2018-05-11:
Great! Will add to the wiki :)

haplm, 2018-05-11:
Brad, I was going to do the same thing (open an issue with this), and saw you did it already, and even cleaned up our conversation. Thanks man!

Bradw-17, 2018-05-11:
@haplm you deserve the credit! thanks for teaching me

Plasmatree, 2018-05-12:
Installing python and running the .py works on every platform. On Linux you don't even need to install python since it's already there! 
The reason for a win executable is to enable at least for some people a solution without all this.

If you just install anaconda with python 2.7 everything should be there! (and you can also start programming your own stuff, which is awsome! )
https://www.anaconda.com/download/

If you also need an editor, try PyCharm ;)

Bradw-17, 2018-05-20:
The convo on FB was deleted, what were the command lines for updating again?

haplm, 2018-05-21:
```
cd PID-Analyzer
git pull
```

optionally '`pip install -r requirements.txt`'

BRadFPV, 2018-07-03:
Having issues with pandas:
> `bash-3.2$ python PID-Analyzer.py --blackbox_decode ../blackbox-tools/obj/blackbox_decode -l ~/Downloads/BTFL_BLACKBOX_LOG_Amped-Air_20180702_sp4.BBL
> Traceback (most recent call last):
>   File "PID-Analyzer.py", line 8, in <module>
>     from pandas import read_csv
> ImportError: No module named pandas`

Pandas appears to have loaded properly:
> Installing collected packages: numpy, scipy, six, python-dateutil, pytz, pandas, subprocess32, pyparsing, functools32, cycler, matplotlib
> Successfully installed cycler-0.10.0 functools32-3.2.3.post2 matplotlib-2.0.0 numpy-1.11.3 pandas-0.22.0 pyparsing-2.2.0 python-dateutil-2.7.3 pytz-2018.5 scipy-1.0.0 six-1.11.0 subprocess32-3.5.2

I verified that pandas exists:
> ultibots:PID-Analyzer Brad$ pip install pandas
Requirement already satisfied: pandas in /usr/local/lib/python2.7/site-packages (0.22.0)
Requirement already satisfied: python-dateutil in /usr/local/lib/python2.7/site-packages (from pandas) (2.7.3)
Requirement already satisfied: numpy>=1.9.0 in /usr/local/lib/python2.7/site-packages (from pandas) (1.11.3)
Requirement already satisfied: pytz>=2011k in /usr/local/lib/python2.7/site-packages (from pandas) (2018.5)
Requirement already satisfied: six>=1.5 in /usr/local/lib/python2.7/site-packages (from python-dateutil->pandas) (1.11.0)

I suspect this might have something to do with my $PATH:
> bash-3.2$ echo $PATH
/usr/local/opt/python@2/libexec/bin:/opt/local/bin:/usr/bin:/bin:/usr/sbin:/sbin:/usr/local/bin:/opt/X11/bin:/usr/local/CrossPack-AVR/bin

Any direction appreciated - TIA - Brad

Bradw-17, 2019-03-07:
From Kevin on the BF Slack:

"At least on Linux the problem is, that the check in line 365 "if str(np.shape(l))=='(4L, 2L)'" fails since the string seems to be different on Windows than on Linux (and probably Mac too). Changing this line to "if np.shape(l) == (4,2):" should fix the problem and **you'll have the exact same scale as on Windows**."

sjpagna, 2020-03-05:
Hi, in my case, this application not generate png image on tmp folder :-( Any solution ?

Bradw-17, 2020-03-06:
@sjpagna just use PIDToolBox: https://github.com/bw1129/PIDtoolbox

Plasmatree unfortunately is dead.