Plasmatree/PID-Analyzer #18: Error run_analysis(clean_path(log_path), args.name, args.blackbox_decode)

trunglp, 2018-05-20:
Today i problem version 0.5  not runing on macbook ,
 The old version 0.3 is work.  

TRUNGs-MacBook-Pro:PID-Analyzer_old1 root# ./PID-Analyzer.py --blackbox_decode ./blackbox_decode -l /Volumes/DATA2/BetaFlight/PID-Analyzer_old1/BTFL_BLACKBOX_LOG_20180512_171144.BBL 
INFO 2018-05-20 17:34:40,180 PID-Analyzer.py:926: Decoding with '/Volumes/DATA2/BetaFlight/PID-Analyzer_old1/blackbox_decode'
INFO 2018-05-20 17:34:40,180 PID-Analyzer.py:928: PID-Analyzer 0.50
INFO 2018-05-20 17:34:40,180 PID-Analyzer.py:929: Hello Pilot!
Traceback (most recent call last):
  File "./PID-Analyzer.py", line 933, in <module>
    run_analysis(clean_path(log_path), args.name, args.blackbox_decode)
TypeError: run_analysis() takes exactly 4 arguments (3 given)
TRUNGs-MacBook-Pro:PID-Analyzer_old1 root#

Plasmatree, 2018-05-20:
There's a bug in the exe. Fixed it in source already. 
Can you use it in interactive mode?

trunglp, 2018-05-21:
@Plasmatree : I use it in macbook. i don't use windows and not run exe . You can fix it running macbook 
the old version 0.3 is work well. 
Thank you

Plasmatree, 2018-05-21:
Does it still exist, even with the current .py?

trunglp, 2018-05-28:
today i use version 0.51
but the error below

-------------------
TRUNGs-MacBook-Pro:PID-Analyzer lptrung$ ./PID-Analyzer.py --blackbox_decode ./blackbox_decode -l  BTFL_BLACKBOX_LOG_F4-Mini_20180528_185448.BBL
INFO 2018-05-28 19:05:49,596 PID-Analyzer.py:953: Decoding with '/Volumes/DATA2/BetaFlight/PID-Analyzer/blackbox_decode'
INFO 2018-05-28 19:05:49,596 PID-Analyzer.py:955: PID-Analyzer 0.51
INFO 2018-05-28 19:05:49,596 PID-Analyzer.py:956: Hello Pilot!
WARNING 2018-05-28 19:05:49,889 PID-Analyzer.py:910: Ignoring BBL session '/Volumes/DATA2/BetaFlight/PID-Analyzer/tmp/BTFL_BLACKBOX_LOG_F4-Mini_20180528_185448_temp0.BBL', 61B < 500000B.
Decoding log '/Volumes/DATA2/BetaFlight/PID-Analyzer/tmp/BTFL_BLACKBOX_LOG_F4-Mini_20180528_185448_temp1.BBL' to '/Volumes/DATA2/BetaFlight/PID-Analyzer/tmp/BTFL_BLACKBOX_LOG_F4-Mini_20180528_185448_temp1.01.csv'...

Log 1 of 1, start 00:12.286, end 02:03.413, duration 01:51.126

Statistics
Looptime            498 avg           62.1 std dev (12.5%)
I frames    3431   43.0 bytes avg   147559 bytes total
P frames  216094   23.2 bytes avg  5008648 bytes total
E frames       2    9.0 bytes avg       18 bytes total
S frames      22    5.0 bytes avg      110 bytes total
Frames    219525   23.5 bytes avg  5156207 bytes total
Data rate 1975Hz  46424 bytes/s     464300 baud

1 frames failed to decode, rendering 6 loop iterations unreadable. 219526 iterations are missing in total (55563ms, 50.00%)

INFO 2018-05-28 19:05:53,334 PID-Analyzer.py:660: Reading: Log 0
WARNING 2018-05-28 19:05:54,761 PID-Analyzer.py:698: No D[0] trace found!
WARNING 2018-05-28 19:05:54,763 PID-Analyzer.py:698: No D[1] trace found!
INFO 2018-05-28 19:05:54,767 PID-Analyzer.py:351: Processing:
INFO 2018-05-28 19:05:54,768 PID-Analyzer.py:655: roll...   
INFO 2018-05-28 19:05:58,593 PID-Analyzer.py:655: pitch...   
INFO 2018-05-28 19:06:02,083 PID-Analyzer.py:655: yaw...   
./PID-Analyzer.py:114: RuntimeWarning: invalid value encountered in divide
  pidin = gyro + pval / (0.032029 * pidp)       # 0.032029 is P scaling factor from betaflight
/usr/local/lib/python2.7/site-packages/numpy/core/_methods.py:26: RuntimeWarning: invalid value encountered in reduce
  return umr_maximum(a, axis, None, out, keepdims)
./PID-Analyzer.py:98: RuntimeWarning: invalid value encountered in less_equal
  low[low <=threshold] = 1.
./PID-Analyzer.py:99: RuntimeWarning: invalid value encountered in greater
  low[low > threshold] = 0.
/usr/local/lib/python2.7/site-packages/numpy/lib/function_base.py:991: RuntimeWarning: invalid value encountered in greater_equal
  not_smaller_than_edge = (sample[:, i] >= edges[i][-1])
/usr/local/lib/python2.7/site-packages/numpy/core/_methods.py:29: RuntimeWarning: invalid value encountered in reduce
  return umr_minimum(a, axis, None, out, keepdims)
/usr/local/lib/python2.7/site-packages/numpy/lib/function_base.py:780: RuntimeWarning: invalid value encountered in greater_equal
  keep = (tmp_a >= first_edge)
/usr/local/lib/python2.7/site-packages/numpy/lib/function_base.py:781: RuntimeWarning: invalid value encountered in less_equal
  keep &= (tmp_a <= last_edge)
INFO 2018-05-28 19:06:05,408 PID-Analyzer.py:546: Making PID plot...
Traceback (most recent call last):
  File "./PID-Analyzer.py", line 960, in <module>
    run_analysis(clean_path(log_path), args.name, args.blackbox_decode, args.show)
  File "./PID-Analyzer.py", line 916, in run_analysis
    test = BB_log(log_file_path, plot_name, blackbox_decode, show)
  File "./PID-Analyzer.py", line 758, in __init__
    self.figs = self._csv_iter(self.heads)
  File "./PID-Analyzer.py", line 775, in _csv_iter
    analysed = CSV_log(h['tempFile'][:-3]+'01.csv', self.name, h)
  File "./PID-Analyzer.py", line 354, in __init__
    self.fig_resp = self.plot_all_resp([self.roll, self.pitch, self.yaw])
  File "./PID-Analyzer.py", line 560, in plot_all_resp
    plt.ylim([-tracelim*1.1, tracelim*1.1])
  File "/usr/local/lib/python2.7/site-packages/matplotlib/pyplot.py", line 1598, in ylim
    ret = ax.set_ylim(*args, **kwargs)
  File "/usr/local/lib/python2.7/site-packages/matplotlib/axes/_base.py", line 3429, in set_ylim
    bottom = self._validate_converted_limits(bottom, self.convert_yunits)
  File "/usr/local/lib/python2.7/site-packages/matplotlib/axes/_base.py", line 3040, in _validate_converted_limits
    raise ValueError("Axis limits cannot be NaN or Inf")
ValueError: Axis limits cannot be NaN or Inf
TRUNGs-MacBook-Pro:PID-Analyzer lptrung$

Plasmatree, 2018-05-28:
It still plotted, didn't it? 
Can you post the log file so i can reproduce it?
What do you mean with 'window size not fix'?

trunglp, 2018-05-29:
this is link log 

https://drive.google.com/file/d/1FWiqzPy4Id0Jz6PKx-2CSKmcNWOuguBU/view?usp=sharing

The windows size not fix size, i use 1280x800, you can see the images

[image]