Plasmatree/PID-Analyzer #19: [Feature request]: add response time info

McGiverGim, 2018-05-25:
I don't know if it is feasible to add the time that the quad needs to arrive from 0 to 1 in the strength vs response time graphs, for both curves. More or less we can now the value, but to compare two graphs a precision number will be better. Maybe next to the graph and not inside?

Thanks for your great app!!!!

[image]

Plasmatree, 2018-05-25:
I've already considered this, but not properly implemented yet. Just getting the crossing point works, but is probably not the best. Furthermore I do not want people to chase a single number to the third digit, but to look at the whole picture that might be a bit more complex.
In your case for example there seems to be an underlying feedback/resonance at maybe 30-40hz. Due to the periodic/continious nature of this superposed signal, the result can have an "instantaneous" component which is not real. In your case visible as this first "bump" in the blue line. >500 there is no feedback happening (or relatively small) and thus the bump is also missing.

Thanks for the input, it's definetly on my list! But I have to do some more tests and and come up with a satisfying solution first. :)

McGiverGim, 2018-05-25:
I think I have too much D, that is the culprit of this line aspect. But my not too powerful motors and heavy props need it to stop the rolls in a clean way. 

Thanks for your feedback!

closed 2018-06-07 by Plasmatree