Plasmatree/PID-Analyzer #24: Tricopters use Yaw D-Term

ResinSmoker, 2018-06-07:
Please show the Yaw D-Term for those flying Tricopters.

Thank you!

Zuldan, 2018-06-07:
About 2 years ago Betaflight removed D on yaw. Butterflight has brought it back. Thanks for the great job you're doing Flo. Looking forward to the JB tutorial on this.

https://www.rcgroups.com/forums/showthread.php?2953533-Betaflight-PID-Analyzer&p=39736387#post39736387

ResinSmoker, 2018-06-11:
A few things i forgot to mention...
-Tricopters use a lot of D-Term to smooth out the Servo / Yaw response. 
-Getting the Servo tuning right is a matter of balancing between the Yaw and Pitch PID's as apparently they play off each other as the rear motor is both responsible for Yaw and Pitch.  The Pitch I-Term in particular is very difficult to adjust in. I've been trying for the last week or so with some measure success but realize that it could be much better.

**Here is a screenshot of my last in-house tune (4S-LiPo)** https://flic.kr/p/Js4Qdx
**Log File** [LOG00001.zip](https://github.com/Plasmatree/PID-Analyzer/files/2089395/LOG00001.zip)
**Diff**
# Triflight on Betaflight 3.2.2 v2 / MATEKF405 (MKF4) 3.2.2 Dec 31 2017 / 15:24:22 (30f68c8) MSP API: 1.36

# resources
resource MOTOR 1 C08
resource MOTOR 2 C09
resource MOTOR 3 C07
resource MOTOR 4 NONE
resource MOTOR 5 NONE
resource MOTOR 6 NONE
resource SERVO 1 C06
resource SERIAL_TX 11 A15

# servo
servo 5 1141 2093 1563 100 -1

# feature
feature DYNAMIC_FILTER

# map
map TAER1234

# serial
serial 30 32 115200 57600 0 115200

# aux
aux 0 0 0 1800 2100
aux 1 1 2 900 1300
aux 2 2 2 1300 1700
aux 3 28 2 1700 2100
aux 4 37 1 1700 2100

# adjrange
adjrange 0 0 3 900 1175 9 4
adjrange 1 0 3 1400 1600 10 4
adjrange 2 0 3 1800 2100 11 4

# master
set gyro_sync_denom = 2
set gyro_lowpass_hz = 70
set gyro_notch1_hz = 0
set gyro_notch2_hz = 0
set max_check = 1950
set blackbox_p_ratio = 128
set dshot_idle_value = 500
set motor_pwm_protocol = DSHOT1200
set failsafe_off_delay = 30
set failsafe_throttle = 1100
set failsafe_kill_switch = ON
set failsafe_procedure = AUTO-LAND
set align_board_roll = 180
set bat_capacity = 1000
set ibata_scale = 179
set servo_center_pulse = 1520
set servo_pwm_rate = 250
set servo_lowpass_hz = 40
set tri_tail_motor_thrustfactor = 89
set tri_tail_servo_speed = 264
set tri_servo_feedback = RSSI
set pid_process_denom = 1
set debug_mode = NOTCH
set dterm_lowpass_type = PT1
set dterm_notch_hz = 0
set vbat_pid_gain = ON
set anti_gravity_gain = 3000
set setpoint_relax_ratio = 1
set p_pitch = 26
set i_pitch = 24
set d_pitch = 43
set p_roll = 34
set i_roll = 20
set d_roll = 33
set p_yaw = 75
set i_yaw = 20
set d_yaw = 140
set rc_expo = 20
set rc_expo_yaw = 20
set roll_srate = 82
set pitch_srate = 82
set yaw_srate = 82
set tpa_rate = 0