Plasmatree/PID-Analyzer #25: Closes instantly

jameyg03, 2018-06-13:
This worked not problem an hour ago but now when I try to open it closes instantly. I did the following in cmd and got the below error. Any ideas?

C:\Users\i\Desktop\Quad stuff\BlackBox Tools>PID-Analyzer_0.52.exe
Failed to write all bytes for Cython.Compiler.Code.pyd

Plasmatree, 2018-06-15:
How much space is left on that drive? 
Any antivirus intercepting?

whakru, 2018-11-08:
Each time you run PID-Analyzer, it creates temporary files for this path: "C: \ Users \ USERNAME \ AppData \ Local \ Temp"
What removes memory on drive C!
He himself does not clean them ... I have to clean myself.
---------
При каждом запуске PID-Analyzer он создает временные фалы по этому пути : "C:\Users\ИМЯПОЛЬЗОВАТЕЛЯ\AppData\Local\Temp"
Что убирает память на диске C!
Сам он их не чистит... Приходиться чистить самому.