Plasmatree/PID-Analyzer #26: Possible feature request

zytra, 2018-06-15:
Hey,

First let me say thank you for this pretty awesome piece of software. I have been using for several months and I am glad to see it gaining interest lately. 

I've personally been using the PID analyzer to analyze vibration more than PID tune and found that to some extend it was possible even though it wasn't really designed for it. The new filter/noise graphs are very helpful and pretty intuitive to analyze.

Back to the PID graphs, I like how you separated the slow and fast input on the bottom graphs, and I am wondering if something similar could be done (to this graph or new ones) in order to help tuning TPA. I know you recommend to turn off as many parameters as possible for the initial tune and I get that. But I am wondering if overlaying several gyro responses for different throttle value could help figuring out some of the vibrations caused at various throttle ranges.

I haven't looked at your code but I suppose it should be possible to input a throttle range in the command line such as: Min Throttle, Max Throttle. With default set to 1000 and 2000, that way someone who doesn't care about this could get the analysis done on the full range, and those who are trying to refine the tune around a specific throttle range could be able to set their min and max and get an analysis that is more focused around this range.

I definitely like the direction you're heading with the new Noise graphs. While I feel like PID tuning is probably easier now than before (thanks to Plasmatree and improvements of the firmware/PID loops, etc) some vibrations due resonance, or various mechanical problems can be tricky to solve, and good analysis tools can go a long way.

Cheers,

Plasmatree, 2018-06-15:
It is already there. The colorplots show response vs throttle (at least where there is good data). 
[image]
If there are single, isolated, jaggy lines, don't believe them. There was too little data and thus not enough averaging.
Any pereiodic resonance at high throttle will also be visible in the noise plots (midthrottle oscillations/propwash often as a band at 20-40hz).

DusKing1, 2018-06-15:
Awesome！！ But I can't understand what those colorplots' color mean.