Plasmatree/PID-Analyzer #27: pid analyzer cmd dissapears

gmohr001, 2018-06-15:
Help needed. Installed pid-analyzer as per Joshua bardwell video but cmd box only flashes on and off when I run PID-Analyzer_0.52.exe.
Nothing else happens
Running win 10 64 bit

Plasmatree, 2018-06-15:
Got blackbox_decode.exe in the same folder?

Stop it from closing by first opeing cmd, and run it from there: 
cd to the location or use "shift+rightclick" (in that folder), select "open powershell" and run ./PID-Analyzer_0.52.exe

gmohr001, 2018-06-15:
Thank you for getting back to me.
Still gives error.
I am sending screenshot showing directory and shell.

Tried to use .\pid-analyzer_0.52.exe as suggested by the prompt but still
gives error

Tx
Geoff

gmohr001, 2018-06-15:
Yup same folder.
tried it from powershell but game this
pid-analyzer_0.52.exe : The term 'pid-analyzer_0.52.exe' is not recognized as the name of a cmdlet, function, script
file, or operable program. Check the spelling of the name, or if a path was included, verify that the path is correct
and try again.
At line:1 char:1
+ pid-analyzer_0.52.exe
+ ~~~~~~~~~~~~~~~~~~~~~
    + CategoryInfo          : ObjectNotFound: (pid-analyzer_0.52.exe:String) [], CommandNotFoundException
    + FullyQualifiedErrorId : CommandNotFoundException

Suggestion [3,General]: The command pid-analyzer_0.52.exe was not found, but does exist in the current location. Windows PowerShell does not load commands from the current location by default. If you trust this command, instead type: ".\pid-analyzer_0.52.exe". See "get-help about_Command_Precedence" for more details.

Then I tried .\pid-analyzer_0.52.exe but then got this
Cannot open self C:\Users\geoff\Downloads\blackbox-tools-0.4.3-windows\PID-Analyzer_0.52.exe or archive C:\Users\geoff\Downloads\blackbox-tools-0.4.3-windows\PID-Analyzer_0.52.pkg