Plasmatree/PID-Analyzer #29: [Feature request] Allow selection of time interval per log to be analysed

Saxin, 2018-06-17:
Flo Hi,

Thanks for the amazing tool, your work is highly appreciated.
Wanted to check is possible to add option to select time interval for running analysis or providing skip X from beginning + skip Y from end to allow eliminate takeoff / landing + allow analyzing specific part of the log.

Thanks

Plasmatree, 2018-06-17:
That requires some major modifications to the interface (to make it usable). As I update that, it will be easier to do so.

"Crashes" should not affect the response analysis as long as there is enough good data.
The noise analysis already ignores the last 2s of each log, which should be sufficient to ignore a crash landing.
I'm also thinking about an auto detection feature for masking in-flight crashes.

Which use case, requiring this feature, is on your mind?

Saxin, 2018-06-19:
Flo thanks for the response,
I think sometimes also one wants to check response of specific "trick" / flight sequence - one example is fast, flat 360 yaw spins - so it can be nice to be able to point plasmatree to specific duration out of a log vs. analyse of full file.

TheLawQuadz, 2018-06-20:
Btw a GUI could be really awesome ;)

theSkonk, 2018-06-21:
Ohh yeah.. my crash affected my log it was. Motor desync and she went into a 3axis spin